markondej/fm_transmitter #2: Stereo support

SandPox, 2015-06-19:
Can you make this support stereo? or fix https://github.com/ChristopheJacquet/PiFmRds to make it work on Pi 2?

markondej, 2015-06-19:
DMA controller was changed in RPI2, so old solutions based on pifm project won't work.

I will add stereo support in fm_transmitter soon.

SandPox, 2015-06-19:
Ok, thank you, I don't know how but I made some change in address by this commit: https://github.com/SandPox/PiFmRds/commit/fa11d6cc122c7d86145408e12606260a43fde152
sometime it work, sometime not but when it work then it less interferes than your code but the chance it work are small (~20 times try I got 1 working, the else are fail to reverse address 0 or some random address), not sure how those code working

markondej, 2015-06-19:
Does it work first time after reset of CPU? Maybe its matter of resetting DMA, PWM or GPIO settings. I don't have RPi to test it right now.

Must say, I'm suprised it works at all ;)

SandPox, 2015-06-19:
sometime work sometime not BUT: it still does transmit silent after exit with error "fail to reverse address.."

markondej, 2015-06-26:
It's because my code switches pin 4 (antena output) back to GPIO operating mode, so no more radio waves are emited. Nothing really important.

In the PiFmRds, part of code which generates errors such as yours is commented with "FIXME" disclaimer. I believe its known error.

What version of raspian are you using?

SandPox, 2015-06-26:
I'm using rapbian jessie (net-installer) with latest kernel

markondej, 2015-06-29:
I've got PiFmRds compiled using your repo https://github.com/SandPox/PiFmRds. On my older raspbian (3.18.7) it works pretty well. It doesn't crash so often but problem is sound quality. Sound was fabulous when I first tried PiFm (by Oliver Mattos and Oskar Weigl) on my RPi 1. With fm_transmitter sound quality is poor as there is no DMA support, and samples transfer isn't so fluent.

But PiFmRds is even worse - a lot of cracks.

SandPox, 2015-06-29:
I don't know but on my side, that repo have better sound quality than your fm_transmitter, less interferes and sound more clear