markondej/fm_transmitter #5: FC in stdin mode

Sg4Dylan, 2015-07-31:
When I use stdin to read audio data, program carsh frequently.
If memory fulled by cache, it crash more often.
Like this:
sox -t mp3 aaa.mp3 -r 44100 -c 1 -b 16 -t wav - | sudo ./fm_transmitter - 77.0
mpg123 -m -C -q -s aaa.mp3 | sudo ./fm_transmitter - 77.0

Could you please fix this problem? 
Thanks.

markondej, 2015-08-03:
Which RPi board and OS are you using?

Sg4Dylan, 2015-08-03:
Sorry, I forgot to say that...

pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux
pi@raspberrypi ~ $ head -n 1 /etc/issue
Raspbian GNU/Linux 7 \n \l

markondej, 2015-08-07:
Please try new release, cache size was restricted to 2MB

Sg4Dylan, 2015-08-10:
It seems OK.
Thanks again!

closed 2015-08-10 by Sg4Dylan