markondej/fm_transmitter #6: How to compile?

meaz, 2015-08-04:
Hi,

I'm a complete Noob! However, I'm really interested in the fm_transmitter as I just got a raspberry pie 2 and would love to try a radio with my pupils. You wrote that to compile this project I can use Code::Blocks IDE (http://codeblocks.org/)... But I don't know how to do so... I downloaded Code Block... And then? What do I need to do?

Thanks a lot.

SandPox, 2015-08-05:
open codeblock, open project, build->build

meaz, 2015-08-05:
Thanks! I opened "fm_transmitter.cbp"; clicked on build then build, but I have an error.

`
||=== Build: Debug in fm_transmitter (compiler: GNU GCC Compiler) ===|`
`C:\Users\Meaz\Desktop\fm_transmitter-master\fm_transmitter-master\stdin_reader.h|54|error: 'pthread_t' does not name a type|`
`||=== Build failed: 1 error(s), 0 warning(s) (0 minute(s), 0 second(s)) ===|`

Did I do something wrong?

Thx

markondej, 2015-08-05:
You are trying to compile it on your Windows computer. As codeblocks works both on windows and linux based OS, fm_transmitter can be compiled only under unix (linux).

You can make compilation right on your RPi device, using raspbian system:
1. Connect your RPi to internet via wifi dongle or via ethernet port
2. Type "sudo apt-get update" in console
3. Type "sudo apt-get install codeblocks codeblocks-contrib" in console, confirm action by typing "Y"
4. Switch to graphical user interface mode (i.e. byt typing "startx") in console
5. In menu (right upper corner of screen) select programming->codeblocks
6. Wait until codeblocks will be ready (takes some time on RPi 1), then open fm_transmitter.cbp and use build->build.
7. Your ready to use binary should be in fm_transmitter/bin/Debug location

meaz, 2015-08-06:
It worked like a charm! Thanks a lot! I can't wait to try with my pupils!

markondej, 2015-08-06:
sudo ./fm_transmitter [filename] [frequency] works only with plain-encoded WAVE files. In order to play mp3 etc. please use stdin streaming (like in second example, with use sox) ;)

codeskyblue, 2015-08-07:
How about add a Makefile. I don't have a codeblocks  installed(too hard for me)

markondej, 2015-08-07:
Looks like I won't escape from makefiles. I've added your makefile

closed 2015-08-07 by markondej