markondej/fm_transmitter #10: What did you fix to get it to work on the pi 2?

skyler440, 2015-08-29:
Hello, this is an issue on another script called PiFox (a Narrow band FM transmitter for the 2 meter ham band for transmitter locating) , which is written for the raspberry pi, but does not work on the pi2.

What did you do to get the signal to transmit on the pi 2? I am in high school and new at programming and am wondering if you could lead me in the right direction for getting pifox to work on the 2. The contacts at pifox end are all dead, and I have tried to email several times. I hope to get it working in the next few weeks because I started an amateur radio club at my high school and I want to have a foxhunt. 

Thanks,
Skyler

SandPox, 2015-08-30:
"DMA controller was changed in RPI2, so old solutions based on pifm project won't work."

markondej, 2015-09-02:
Like SandPox wrote, DMA controller support has changed in RPi2 so it's not that simple.

You can try to clone: https://github.com/fotografAle/NBFM, then edit nbfm.c file, on line 168:
```
 0x20000000 //base
```
to:
```
 0x3F000000 //rpi2 base
```
then compile nbfm:
```bash
gcc -o3 -lm -std=c99 -o nbfm nbfm.c
```
This will result with creating executable file nbfm, which you should copy into your pifox location (overwriting existing one). If you're lucky it could work.

skyler440, 2015-09-04:
Thanks, I tested it and it is transmitting a carrier now, but no audio is working, and it is very buggy (ex. found the carrier way off frequency at one time)

Here is my live progress on getting Nbfm or pifox working and everything I have tried so far

https://docs.google.com/document/d/1j_9KvkxA1LSg185Ded42PzbSZbsrGQTyy5o0kZxz9GM/edit?usp=sharing

iliveforgod347, 2016-08-14:
how would you go about doing this on rpi3?  I would prefer a solution that does not use so much cpu power.  Thanks!