markondej/fm_transmitter #12: does not transmit for duration of file

ronlaws86, 2015-09-10:


markondej, 2015-09-11:
Can you provide us some more info? What kind of file is it? Are you using stdin streaming or playing wave file direct?

ronlaws86, 2015-09-11:
Wave file directly and stdin both exhibit the same odd behaviour, using
provided wav file or providing a different file yields the same result, the
carrier wave (silence) can be heard, followed by the audio, then after
about 5 or more seconds (sometimes may be up to 8) the carrier wave
disappears and the receiving radio returns back to static

ronlaws86, 2015-09-12:
Sorry to answer my own post, but i found out what the problem is, i'll post it in case anyone else comes across this: 

I had a kernel module loaded from playing around with a Digital thermal sensor, which happens to use the same pin for the data input from the sensor, which must have been repeatedly probing the gpio pin and re-setting it's direction from Out to In, which was interrupting the transmitter. 

entering: 
sudo rmmod w1-gpio
sudo rmmod w1-therm
before running fm_transmitter solved the problem, and the broadcast lasted for the duration of the provided wav file this time.

closed 2015-10-21 by markondej