markondej/fm_transmitter #17: Pi 2 Compile Error

Addicted5, 2015-11-20:
pi@raspberry ~/Music/fm_transmitter $ make
gcc -Wall -fexceptions -lpthread -lm -O3 -fpermissive -fno-strict-aliasing -c main.cpp
gcc -Wall -fexceptions -lpthread -lm -O3 -fpermissive -fno-strict-aliasing -c wave_reader.cpp
gcc -Wall -fexceptions -lpthread -lm -O3 -fpermissive -fno-strict-aliasing -c stdin_reader.cpp
stdin_reader.cpp: In constructor ‘StdinReader::StdinReader()’:
stdin_reader.cpp:57:91: warning: invalid conversion from ‘void (*)(void*)’ to ‘void* (*)(void*)’ [-fpermissive]
     int returnCode = pthread_create(&thread, NULL, &StdinReader::readStdin, (void*)&params);
                                                                                           ^
In file included from /usr/include/arm-linux-gnueabihf/c++/4.9/bits/gthr-default.h:35:0,
                 from /usr/include/arm-linux-gnueabihf/c++/4.9/bits/gthr.h:148,
                 from /usr/include/c++/4.9/ext/atomicity.h:35,
                 from /usr/include/c++/4.9/bits/basic_string.h:39,
                 from /usr/include/c++/4.9/string:52,
                 from error_reporter.h:37,
                 from stdin_reader.h:40,
                 from stdin_reader.cpp:34:
/usr/include/pthread.h:244:12: note: initializing argument 3 of ‘int pthread_create(pthread_t*, const pthread_attr_t*, void* (*)(void*), void*)’
 extern int pthread_create (pthread_t *__restrict __newthread,
            ^
gcc -Wall -fexceptions -lpthread -lm -O3 -fpermissive -fno-strict-aliasing -c error_reporter.cpp
gcc -Wall -fexceptions -lpthread -lm -O3 -fpermissive -fno-strict-aliasing -c transmitter.cpp
transmitter.cpp: In member function ‘void Transmitter::play()’:
transmitter.cpp:134:90: warning: invalid conversion from ‘void (*)(void*)’ to ‘void* (*)(void*)’ [-fpermissive]
     int returnCode = pthread_create(&thread, NULL, &Transmitter::transmit, (void*)&params);
                                                                                          ^
In file included from /usr/include/arm-linux-gnueabihf/c++/4.9/bits/gthr-default.h:35:0,
                 from /usr/include/arm-linux-gnueabihf/c++/4.9/bits/gthr.h:148,
                 from /usr/include/c++/4.9/ext/atomicity.h:35,
                 from /usr/include/c++/4.9/bits/basic_string.h:39,
                 from /usr/include/c++/4.9/string:52,
                 from wave_reader.h:37,
                 from transmitter.h:37,
                 from transmitter.cpp:34:
/usr/include/pthread.h:244:12: note: initializing argument 3 of ‘int pthread_create(pthread_t*, const pthread_attr_t*, void* (*)(void*), void*)’
 extern int pthread_create (pthread_t *__restrict __newthread,
            ^
transmitter.cpp: In destructor ‘virtual Transmitter::~Transmitter()’:
transmitter.cpp:222:35: warning: invalid conversion from ‘volatile void*’ to ‘void*’ [-fpermissive]
     munmap(peripherals, 0x002FFFFF);
                                   ^
In file included from transmitter.cpp:40:0:
/usr/include/arm-linux-gnueabihf/sys/mman.h:76:12: note: initializing argument 1 of ‘int munmap(void*, size_t)’
 extern int munmap (void *__addr, size_t __len) __THROW;
            ^
g++ -Wall -fexceptions -lpthread -lm -O3 -fpermissive -fno-strict-aliasing -o fm_transmitter main.o wave_reader.o stdin_reader.o error_reporter.o transmitter.o

markondej, 2015-12-07:
these are compilation warnings, nothing really important, I'll try to fix this in nearest release

markondej, 2015-12-16:
Comile warnings fixed in new branch (I will merge it into master after testing)

closed 2015-12-16 by markondej