markondej/fm_transmitter #19: ACCESS(base, offset) macro

Horkyze, 2016-01-17:
I've asked a question on StackOverflow regarding the ACCESS macro definition. I didn't quite understand what it does (I do now). People were nice and pointed out that it could cause undefined behaviour, so I am posting here. Here is the link to the question. 
http://stackoverflow.com/questions/34832622/what-does-this-c-macro-mean

markondej, 2016-01-18:
Yes, indeed it could cause some undefined behaviour but on undefined hardware. This software is targeted only 32-bit RPi ARM processors, so as use of simple types ("unsigned int" instead of "uint32_t") is less elegant, it does the job. We do use typical RPi peripherials here, so even, if code compiles on i.e. i386 platform, there is no point in using it.

I bet this code could be used on other development boards (with minimal changes) and it's open source so feel free to modify it according to your needs ;)

markondej, 2016-02-29:
Since RPi3 uses 64 bit processor, I guess it'd be nice to not use platform independent types, so guys pm stackoverflow are right

markondej, 2019-09-25:
This macro does not exist any more

closed 2019-09-25 by markondej