markondej/fm_transmitter #20: Slow motion audio!!!

someaguy, 2016-02-18:
Hey there! I signed up to this site just to figure out what is going on with an issue I have with your project. 

When I want to transmit audio from the sound card's LINE IN, using the: (arecord -D hw:1,0 -c1 -d 0 -r 22050 -f S16_LE | sudo ./fm_transmitter -f 106.3 -) the audio being broadcasted is very slow motion audio. 

I get a message when I hit that command: Warning: rate is not accurate (requested = 22050Hz, got = 44100Hz)
         please, try the plug plugin 
Playing: stdin, 22050 Hz, 16 bits, mono

If I keep the broadcast going, I eventually get repetitive error messages....

overrun!!! (at least 3.014 ms long)
overrun!!! (at least 3.519 ms long)
overrun!!! (at least 2.239 ms long)
overrun!!! (at least 2.768 ms long)
overrun!!! (at least 2.742 ms long)
overrun!!! (at least 3.364 ms long)
overrun!!! (at least 3.000 ms long)
overrun!!! (at least 2.617 ms long)

I am currently using the C-Media Electronics, Inc. Audio Adapter for the sound device. The OS is Raspbian Jessie (February 2016)

markondej, 2016-02-18:
I'd try modifying the stdin_reader.h file. Check the [line 43](https://github.com/markondej/fm_transmitter/blob/master/stdin_reader.h#L43), there is default sample rate value for stdin stream. Change it to:
`#define STREAM_SAMPLE_RATE 44100`

Then save the file, and rebuild whole project ie:
`make clean`
`make all`

someaguy, 2016-02-19:
Yes! that certainly did the trick! Thank you! :) This may be something note for those who wants to use your project.

closed 2016-02-29 by markondej