markondej/fm_transmitter #21: Error while opening file, data corrupted

bobriguez, 2016-02-22:
Hey,

I noticed that whenever I tried to play a wave file, it returned the error "Error: Error while opening FV.wav, data corrupted". At first, I figured it might be an actual issue with the file and tried different ones, but every single one has returned either "unsupported WAVE format" or the data corrupted error. I check out the files myself and they do appear to be in working order. I also looked into the format and encoding.

First I did a "avconv -i" on the included sample file to find out what format was necessary. It returned:
" Stream #0.0: Audio: pcm_s16le, 22050 Hz, 1 channels, s16, 352 kb/s"
Then I converted multiple files to those exact specifications. I ran the same check on them and it returned the very same values.
"Duration: 00:04:05.52, bitrate: 352 kb/s
 Stream #0.0: Audio: pcm_s16le, 22050 Hz, 1 channels, s16, 352 kb/s"

I even tried separate converters, in case this one was not up to snuff, but I got the same error every time.

Now, I would just use stdin, but the audio quality on that is way lower than when playing the sample audio directly from the file.

markondej, 2016-02-29:
Usualy this error occures when subchunk's "data" and "fmt" cannot be found in standard location in file. Maybe your files aren't PCM wave files (lately wave files uses mp3 codecs, which won't work), so the only option you have is to use stdin. You can increase standard framerate by editing stdin_reader.h file and recompilling whole project (make clean/make all). 

And then using stdin input ie:
```
sox star_wars.wav -r 44100 -c 1 -b 16 -t wav - | sudo ./fm_transmitter -f 100.6 -
```

KDragonM, 2016-03-17:
Not exactly a solution, but if you use [media.io](http://media.io/) to convert your file from mp3 to wav it works without a problem.