markondej/fm_transmitter #24: Does it work with Pi3 ?

pcotret, 2016-03-12:
http://www.icrobotics.co.uk/wiki/index.php/Turning_the_Raspberry_Pi_Into_an_FM_Transmitter

This version does not work "out of the box". I assume PWM, DMA or clocks have a different behavior in BCM2837 but as there is no documentation on this chip yet...

markondej, 2016-03-14:
As this project works fine with RPi2, it should also work with RPi3. I dont't have RPi3 yet, so cannot confirm for 100%.

pcotret, 2016-03-14:
OK, I used the ICrobotics version, but it doesn't work without 
modifications... (I'm afraid that addresses of DMA/PWM may have changed 
in BCM2837 but there's no datasheet yet :-( ).

I'll try your version tonight (even if your README says that Pi3 works)

KDragonM, 2016-03-17:
I can confirm it works for the Pi 3 as I just tested it and it works.

pcotret, 2016-03-17:
I forgot to answer but yes, it works ! (pretty fine, sounds was clear 
and quite powerful)
(figure : transmission disabled/enabled, in the French commercial FM band)

closed 2016-06-20 by markondej