markondej/fm_transmitter #28: very weak signal.

jaxtheop, 2016-04-04:
My signal is very week and full of static. I have a 30-40 cm antenna plugged into GPIO4. I noticed on this guys video of this project (https://www.youtube.com/watch?v=h2Z0s7_z6IM) at 0:19, I see he has stdin_reader.h and transmitter.cpp and readme.md as green (Executable or recognized data file) but when i use the command ls, my files are not green like his, only my transmitter_fm is green. Does this have something to do with the poor quality of the signal?

markondej, 2016-04-04:
File color presented by console has nothing to signal range. Please give us some more details about your setup (RPi 1 B/B+, RPi 2/3 ???) and write what do you mean by "signal is very weak"? Normally I was able to get it working in 10m range.
Additionaly chceck you anthena connection for cold joints, or try to use other anthena (try using thin wire).

jaxtheop, 2016-04-04:
I have the rpi 3, and by weak signal, i mean that from a foot away i can barely make out the star wars song in the static. Im new to this, how would i check for cold joints? I have also tryed a 20cm female jumper cable and it doesent really improve any, gets a little worse

Thanks

markondej, 2016-04-04:
I don't have RPi 3 yet, so maybe someone else could help?

KDragonM, 2016-04-13:
I had this issue but it turned out that the antenna was attached to the wrong pin. It's supposed to be connected to the third inner pin counting from the camera serial port. Also, you could try attaching a longer antenna. Hope this helps