markondej/fm_transmitter #29: Can you please put a video tutorial up?

Mikalangelo, 2016-04-29:
I am extremely new to raspberry pi and got this as a gift. I want to learn how to make this project. The words don't really mean anything to me so thats why I want a video tutorial

markondej, 2016-04-29:
https://www.youtube.com/watch?v=h2Z0s7_z6IM

Mikalangelo, 2016-04-29:
What os do you use? NOOBS or Raspbian?

Mikalangelo, 2016-04-29:
What program do you use for the code? Terminal or Python?

markondej, 2016-04-29:
NOOBS is not OS, but simple os-choosing application with graphic layer. You can use NOOBS image to run Raspbian. This project should compile on any linux-based OS, but I recommend Raspbian. Code was written in C++ (2003 standard), so to run it, you have to:
1) Download it - ie. start your RPi with raspbian, establish internet connection (plug in ethernet wire or use WiFi dongle), run Terminal, then type:
`git clone https://github.com/markondej/fm_transmitter`
2) Compile it - as C++ code is human readable, in order to be run on computer it should be translated into CPU machine code. Usualy this "translation" is called "compilation". To compile fm_transmitter you will need to change current working directory to where you downloaded it (in terminal):
`cd fm_transmitter`
and then use make command to perform compilation
`make all`
3) Tada - new file should be generated - fm_transmitter, containing machine code runable on your RPi. Test it with use of this command in terminal:
`sudo ./fm_transmitter -f 100.0 starwars.wav`

Mikalangelo, 2016-04-29:
Thank you I'll try it as soon as possible!

markrsmith, 2016-06-05:
Thank you for the awesome directions! Quick type correction though, instead of `sudo ./fm_transmitter -f 100.0 starwars.wav`you need to type `sudo ./fm_transmitter -f 100.0 star_wars.wav`

closed 2016-06-20 by Mikalangelo