markondej/fm_transmitter #34: high cpu usage

xcommanderx, 2016-06-11:
I'm using fm_transmitter with RPi3, its works fine! thx!

I tried to use with sox, ffmpeg, mp123 and vlc, and always fm_transmitter use ~115% of CPU, something like 100% of one core full time.

this behavior is normal?

markondej, 2016-06-13:
Yes, this might be normal behaviour. fm_transmitter uses software way to distribute proper frequency in proper moment to antenna output, so it does consume almost whole processing power. Currently I'm starting to work on new version with DMA and PWM peripherals support, so it will work more like https://github.com/ChristopheJacquet/PiFmRds

markondej, 2019-01-07:
DMA support added, CPU usage should be lower

closed 2019-01-07 by markondej