markondej/fm_transmitter #39: Intallation Instruction?

Sryptic, 2016-10-04:
The installation instructions in the read me are really vague. I've tried everything I can think of but I can't  seem to install this :( 

I'm on a Pi2+B, how would I install this from the terminal?
anyone able to help?

Sryptic, 2016-10-04:
So I've managed to get it installed and somewhat working(haven't actually tested any FM output yet).
When I run the command  `sox star_wars.wav -r 22050 -c 1 -b 16 -t wav - | sudo ./fm_transmitter -f 100.6 -` It shows
-bash: sox: command not found
Playing: stdin, 22050 Hz, 16 bits, mono

What does the 'stdin' mean? and why is there a 'sox' command not found error?

markondej, 2016-10-04:
fm_transmitter has simple built in wave file support, just type:
`sudo ./fm_transmitter -f 100.0 star_wars.wav`

To support more sophisticated file formats (mp3) you can use ie. sox. Sox is not part of this project, so you have to install it via apt-get, by typing:
`sudo apt-get install sox`
Then you can install sox mp3 addin:
`sudo apt-get install libsox-fmt-mp3`

closed 2016-10-21 by markondej