markondej/fm_transmitter #42: Can not change the frequency

Tunguska12, 2016-11-16:
Hi guys! 
Currently, I can not change the broadcast frequency.
First, I used command  `sudo ./fm_transmitter [-f 100] [-r] filename ` to play a song in 100Mhz then I ctrl+C to stop it.
I used another command `sudo ./fm_transmitter [-f 85.5] [-r] filename` to play the song in different frequency but it did not change. The song still stayed in old frequency 100Mhz.

What I am supposed to do guys? I'm just a newbie in Linux as well as Raspberry. Any suggestion would be appriciated.

markondej, 2016-11-16:
Hi,
just don't write square brackets (they mean something optional)

Tunguska12, 2016-11-16:
Yeah it worked! thanks a lot :D
and I want to ask just one more question, what I have to do if I want to play all songs in a folder inside /fm_transmitter.

markondej, 2019-01-07:
You can add multiple files and even loop them, please see issue: #56

closed 2019-01-07 by markondej