markondej/fm_transmitter #43: interference and so lower volume

kensukekita, 2016-11-23:
hello,
i connected the antena to the gpio 4 pin and when i'm transmitting i got so much interference and low volume

i tried with lot of frequencies but all what i listen to is statics interference and low volume it's hard to listen the song tat i'm transmitting

i used this command

`ffmpeg -i opus.mp3 -f s16le -ar 22.05k -ac 1 - | sudo ./fm_transmitter -f 102.9 -`
and tried this one

`sox 1.wav -r 22050 -c 1 -b 16 -t wav - | sudo ./fm_transmitter -f 102.9 -`

but none works(both works but with the problems that i already told in this topic)

diaphon, 2016-11-24:
did you try to push the volume with `-v 10` ?
`sox -v 10 1.wav -r 22050 -c 1 -b 16 -t wav - | sudo ./fm_transmitter -f 102.9 -`
worked for me (see #40 )

kensukekita, 2016-11-25:
I will try thank you
I will reply if it worked

kensukekita, 2016-11-25:
it didn't worked i still listening the stactic and interference :c

kensukekita, 2016-11-25:
i just noticed that when i tried to transmit on 104.9 it transmit on 105.1 i dont know why it happens :c

kensukekita, 2016-11-28:
i just reinstalled raspbian and it is working now! thanks @diaphon

kensukekita, 2016-11-28:
yes it is working pretty good now thanks for your help :)

diaphon, 2016-11-28:
good to hear ;-)