markondej/fm_transmitter #48: Is there a way to transmit on every fm station

connwat, 2017-01-27:
Is there a way to transmit on every fm radio station using the usb sound card command

haddockdev, 2017-02-10:
Unless your somewhere's where nobody is broadcasting on the FM band (highly unlikely) you really shouldn't do it. 
Plus, I would think it would have a very low chance of working.

n1iic, 2020-05-07:
This is done on every FM broadest channel already. Years ago this is how elevator music was distributed throughout cities, and currently it's how traffic info is sent to navigation systems,  RDS and HD radio are also part of this. 
Inside the FM signal are sub carriers with all of this extra info,  but it's done in a sophisticated way. With 110 channels in the FM band, I don't believe there's enough processing power in a pi to be able to have 110 different carriers of even the same program. In analog you could have the same 100 kHz signal repeat, though. 
But I agree with ghost, not a good idea unless

markondej, 2021-03-16:
There is no way of transmitting on more than two frequencies with one RPi board at same time. Question repeated at #135

closed 2021-03-16 by markondej