markondej/fm_transmitter #51: Why playback cuts after few seconds [EXPLAINED]

RichMStudios, 2017-05-17:
When you convert anything with media.io WINDOWS APP, it doesn't fully convert - that is why it does not play it all. 
However, it works when you convert it with media.io web converter. Also, you can convert it with Audacity - it works too.

vettorazi, 2017-09-02:
here it's not playing more than 5 seconds. not even the star wars song that came with it.

RichMStudios, 2017-09-02:
Sorry, maybe only someone has these problems

ZachDunham, 2017-10-05:
I'm still having this issue unfortunately on a b+. The transmitter intermittently seems to cut out.  Rebooting the pi sometimes helps, but not always. I'm wondering if calling GPIO.cleanup() would help.  Anyone figure out how to solve this issue?