markondej/fm_transmitter #53: Changing gpio pin

Noonlord, 2017-07-14:
Can i change gpio pin? What changes i need to make before compiling?

clerie, 2017-09-21:
Probably you can't because "this project uses the general clock output to produce frequency modulated radio communication." And currently the is only one pin wich supports this, I think.

markondej, 2017-09-22:
According to RPi peripherals documentation available here (see Page 102):
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf
there is possibility of using general purpose clock output on GPIO 4, 20, 21, 32, 34, 42, 43, 44.

GPIO layout is available ie. here: https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png

GPIO's 20 and 21 are available on PIN's 38 and 40, which respectively uses CLOCKS #0 and #1.

This project is configured to use CLOCK #0 on GPIO 4 (PIN 7).
To change output GPIO You need to change given GPIO's function (see: https://github.com/markondej/fm_transmitter/blob/master/transmitter.cpp#L211 and Peripherials doc page 90) to serve as CLOCK output (page 102).

In order to change used clock You need to change clock's base address (https://github.com/markondej/fm_transmitter/blob/master/transmitter.cpp#L47).

For example if You want to change output to PIN 40 (GPIO 21) You will have to do some changes in transmitter.cpp:

Change clock address to use CLOCK1 instead of CLOCK0 (line 47):
`#define CLK0_BASE 0x00101070
#define CLK0DIV_BASE 0x00101074`
to:
`#define CLK0_BASE 0x00101078
#define CLK0DIV_BASE 0x0010107c`

Change GPIO 21 function to Alt5 (line 207):
`ACCESS(peripherals, GPIO_BASE) = (ACCESS(peripherals, GPIO_BASE) & 0xFFFF8FFF) | (0x01 << 14);`
to:
`ACCESS(peripherals, GPIO_BASE + 0x00000008) = (ACCESS(peripherals, GPIO_BASE) & 0xFFFFFFC7) | (0x02 << 3);`

That should work...

closed 2020-03-31 by markondej