markondej/fm_transmitter #58: PI crashes when I start transmission

juhssi3, 2017-10-07:
When I start to transmit, the song plays well and entirely, but the GUI goes really weird looking, all USB powered devices are shutted off (including mouse an kboard) and if there is SSH connection, it fails immediately. I have changed the transmitting gpio to pin 40 following your instructions because the default pin is dead.

Tonoanaya89, 2017-11-30:
check your power source, FM radio transmission requires more amps from your raspberry pi. Greetings.

Hoshino-Hikari, 2020-03-10:
I have the same problem, but may be mine is worse, when I start it, everything is like what you said, in addition, no task is activated, evev the task manager (it just stucked), after a while, the transmitter just repeat a piece of audio that is being played (2-3 seconds).

willis936, 2020-12-07:
I had this issue when trying to transmit below 100 MHz.  This code modulates the SoC base clock.  It has a limited range where things do not break.