markondej/fm_transmitter #60: How to increase range?

0xCoto, 2017-11-05:
I had 2 questions:

1) How can I possibly increase the radius range of the transmission? (I.e. with external antennas, external power source etc.)

2) What is the difference between running this on a RPI 0, RPI 0 W, RPI 1, RPI 2 and RPI 3?

clerie, 2017-11-15:
On RP2 I added an antenna with lenght of ~1.5m and reached a range of ~250m in City
The performance may be better, when you use stronger power supplies

0xCoto, 2017-11-18:
Wouldn't increasing the length require more power (i.e. externally powering the antenna)? If so, how is that possible? Would increasing the length even more achieve better range?

And did you use a copper wire or did you somehow implement a "professional" Antenna?

iBug, 2017-11-23:
I'm afraid you won't find an applicable external power supply easily, but as for me, an antenna of 2.0m (6.6ft) gives a surprising range with 200+m, with obstruction.

0xCoto, 2017-12-04:
What kind of wire did you use? Also did you straighten the wire or what? And what frequency did you try?

iBug, 2017-12-12:
@0xCoto Should be 85.0 MHz. Yes I straightened the wire. The 200m can only be reached from a direction that's perpendicular to the wire. If you try to catch the signal from a parallel direction (i.e. the way the wire points to), the range will drop significantly to less than a half of the longest.

Brandon4466, 2017-12-14:
@iBug @clerie What did you make the wire out of? Also, how did you connect it to the Pi?

Thanks for any information you can supply!

KG7UAT, 2018-05-10:
Be very careful when extending range. The Raspberry Pi radio generates really bad harmonics, which basically means you are transmitting on more than one frequency and possibly disturbing government, aircraft, or other transmissions which is illegal. Do NOT extend range without a band-pass filter.