markondej/fm_transmitter #64: Every wave file I put in folder is "corrupted"

JuppyPi, 2018-02-04:
Every file I convert from mp3 to wav using audacity it declared corrupted and even file directly downloaded from wav sources are also corrupted. The only thing that plays is the starwars theme that came with it. Any suggestions?

dudenugget2000, 2018-02-25:
This is happening to me as well.

Noonlord, 2018-06-05:
You can use audacity and export it again. If you download from YouTube you can use this: https://github.com/Noonlord/YouTube-WAV-Downloader/blob/master/YouTube_to_WAV.py

if you convert with ffmpeg add -flags bitexact

metadiablo-glyph, 2020-04-12:
> 
> 
> You can use audacity and export it again. If you download from YouTube you can use this: https://github.com/Noonlord/YouTube-WAV-Downloader/blob/master/YouTube_to_WAV.py
> 
> if you convert with ffmpeg add -flags bitexact

Trying both of these options failed for me.