markondej/fm_transmitter #65: Just play 5 seconds or less

MonsterDeveloper, 2018-02-12:
hey guys, anyone having this same issue?
i send the command `sudo ./fm_transmitter -f 104.2 -r star_wars.wav` and only play for 4 or 5 seconds and goes off. what is happening?

playinmyblues, 2018-02-12:
This seems to be a common issue. I found a solution online somewhere that detailed a short script to turn off the temperature monitoring on the CPU.

markondej, 2018-02-12:
Hi, please specify which RPi board and what kind of OS are you using.

MonsterDeveloper, 2018-02-12:
So, my board is RPi 3 model B and my OS is Raspbian 8.0 Jessie

MonsterDeveloper, 2018-02-12:
@markondej and I am using just jumper cables as my antenna. Like this:
[image]

MonsterDeveloper, 2018-02-12:
If @markondej you need demo, I can send you audio file of video.

MonsterDeveloper, 2018-02-12:
Hmm, interesting thing that I booted into Kali Linux on my RPi, installed make, gcc and g++ and everything worked just fine. It seems that problem with os (Raspbian). WHat do you think it is?

TheMakaf, 2018-05-05:
Was same problem. I used 800 mA power source, thats too low for radio transmitter work. Now I'm using 1.5 A or 2 A power source, everything works.

playinmyblues, 2018-08-14:
The link to solve my problem is here:
https://raspberrypi.stackexchange.com/questions/47893/pifm-rds-cuts-off-after-a-few-seconds

In short, the command is:
sudo rmmod w1-gpio

I notice that somebody else had a problem that was solved using a better power source. I am using an RPi3 with an RPi power source rated for 5V 2.5A.

Given all that, I do not know why it works when I run the command provided above so it is best if you research the problem before using it.

closed 2018-08-14 by MonsterDeveloper