markondej/fm_transmitter #78: Not work with the Pi4

cpnetserver, 2019-07-18:
Hy guys!
No problem with the Raspberry Pi 3B+...
The same O.P. [Debian 10], same configuration, but with the PI 4 not work.
Is there some tips?

kuchajun, 2019-07-25:
I have the same problem.You've solved it now？

silicator, 2019-07-25:
im not sure but i assume that the pi4 has other addresses defined for bases etc like a difference between pi 0,1,2,3 . i found something but not sure if this is correct, maybe someone can test this compiled with a new makro for pi4 and give a report back:
#define PERIPH_VIRT_BASE               (0xFE000000)
#define PERIPH_PHYS_BASE               (0x7E000000)
#define DRAM_PHYS_BASE                 (0xC0000000)
#define MEM_FLAG                       (0x04)
#define XTAL_CLOCK                     (54.0E6)
#define DMA_CHANNEL                    (6)

markondej, 2019-08-20:
@silicator can you please link the source

I obtained my RPi 4 today so I'll try to fix this