markondej/fm_transmitter #80: Not working on Raspberry Pi 4

stk122, 2019-09-08:
I make the setup of FM Transmitter like I do in my old Rpi2 and simply not work. The command start ok, with the message:

_"pi@RPI4stk122:~ $ sudo /home/pi/fm_transmitter/./fm_transmitter -r -b 200 -f 90.6 /home/pi/fm_transmitter/musica/todas.wav"_

_Broadcasting at 90.6 MHz with 200 kHz bandwidth
Playing: /home/pi/fm_transmitter/musica/todas.wav, 44100 Hz, 16 bits, stereo_

But not sound recibed on radio. If I put the same command in Rpi2 works fine. I put a wire in the same GPIO PIN 4, I disable 1-Wire on _raspi-config_, disable another conflictive modules with _"sudo rmmod w1-gpio"_ and _"sudo rmmod w1-therm"_, but it continue not working.

Any way to fix it? I don't know what else to try.

markondej, 2019-09-11:
There is a bug - for RPi 4 peripherals address is set to 0. The bug is Raspberry Pi's inner libraries (in other words: this isn't working as described by RPi foundation https://www.raspberrypi.org/documentation/hardware/raspberrypi/peripheral_addresses.md). I will try to bypass this problem and publish solution as soon as possible (tomorow).

vssj322, 2020-06-23:
Mine is also not working

Missingfiles2021, 2023-02-22:
I found out a solution: 
1.Make sure you overclock your CPU at least above 50 % and it should not be left at idle.
2. Do /sudo rmmod w1-gpio
This might sound the opposite of what the others are saying, but I'm just hoping for an update or a patch from the Raspberry Pi 4 devs to diagnose or fix this problem. As for now I'm speculating that it is a bug.