markondej/fm_transmitter #90: How to play playlists/folders of files?

cyliim, 2019-10-08:
Is there a way to make a folder of WAVE files, and then get the program to choose one and play it, then repeat through the entire list?

jfla15, 2019-11-16:
It is as easy as going `sudo ./fm_transmitter music/*` where 'music' is a folder you've created for this purpose.

joshyyoshy123, 2022-10-12:
On mine It just says data corrupted on the file

markondej, 2022-10-12:
Hi, there is a bug with wildcard names and filenames like "*" are not resolved properly. But as I remember you can strictly list all the files you want to broadcast in program parameters ie.:
```
sudo ./fm_transmitter a.wav b.wav c.wav
```

joshyyoshy123, 2022-10-12:
Thanks I am just stupid and just realised to just put the star on after the name of the file thx anyway