markondej/fm_transmitter #92: run fm_transmitter on startup/headless

vmatthieus, 2019-12-21:
hi markondej,
im using this as a school project and since i won't have access to a computer or a display and keyboard during the demo, is there a way to make the script run automatically on startup?
if not, using other programs, can i make the pi output a wifi AP that i can connect to, and then do everything via ssh on my phone?

thanks in advance!

markondej, 2019-12-22:
To run an executable on startup the simpliest way is to add proper command do /etc/local.rc file, if you want to be more sophisticated register fm_transmitter as an systemd service, see: https://www.linode.com/docs/quick-answers/linux/start-service-at-boot/ In order to make RPi working as an AP please read https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

micfun123, 2021-11-19:
Do you have a prewritten systemd Service?

kzl86, 2022-02-16:
I'm using the following file to describe my service. I'm using a bash wrapper to supply the fm_transmitter with proper parameters, looping through a folder of wav files.
When I stop the service the radio output is not the static noise, which normal, but a mute signal. Starting back up the service is working: I get the sound properly. Only if I'm restarting the raspberry pi solves this behaviour: I get back the static noise, like after a simple playback.

[Unit]
Description=fm_transmitterd.sh service
After=network.target
StartLimitIntervalSec=0

[Service]
Type=simple
Restart=always
RestartSec=1
User=pi
ExecStart=/usr/local/bin/fm_transmitterd.sh

[Install]
WantedBy=multi-user.target

kzl86, 2022-02-19:
I've solved this issue, which was related to linux signals. According to this doc https://www.man7.org/linux/man-pages/man5/systemd.kill.5.html the default signal is SIGTERM when stopping the service. Now when I stop the fm_transmitter from the cli I use CTRL+C, which is SIGINT.

The solution is to overwrite the default signal in the service description. To do this add the following line to the [Service] section:
KillSignal=SIGINT