markondej/fm_transmitter #98: GUI for this project

Alfon786, 2020-01-10:
Would it be possible to create an GUI for this program?
It's for a school project, we would create a radio program for the institute newspaper and the people that will manage the radio don't know how works the command lines.

How could it be done?
I have basic programming knowledge but I'm not sure about what it's the best way for do this.

CodeSarthak, 2020-02-22:
I am all for making a GUI for this project. 

Count me in.

markondej, 2020-03-31:
Assuming that You want to run GUI on same machine that will be transmitting, there are few nice frameworks for linux:
https://wiki.qt.io/Qt_for_Beginners
http://wiki.codeblocks.org/index.php/WxSmith_tutorial:_Hello_world