markondej/fm_transmitter #100: After ^C the GPIO pin still has the GPCLK0 signal active

waterwin, 2020-01-15:
In many other programs I have seen that the program resets the state of the used GPIO when quitting the transmit.

Here it appears that the pin stays used.
pi@raspberrypi:~ $ sudo raspi-gpio get 4
GPIO 4: level=0 fsel=4 alt=0 func=GPCLK0

ldezem, 2020-01-20:
Could this be related to the [issue](https://github.com/markondej/fm_transmitter/issues/96) that`pkill fm_transmitter` gets the program stuck into a loop were it broadcast a fragment of the sound indefinitely?

waterwin, 2020-01-21:
I do not know. I would need to measure the signal on the pin to see that happening. But this appears not to be the case as my receiver is not picking up any signal anymore.
I have seen in other code a good way to end the transmission and free up the GPIO pins,. That is in the code itself and not by killing a running program. I will look this up.

https://github.com/ChristopheJacquet/PiFmRds/blob/master/src/pi_fm_rds.c
lines 242-271 might help for a clean termination

Good luck
ErwinPE3ES

markondej, 2020-03-21:
Fixed this issue in branch https://github.com/markondej/fm_transmitter/tree/rpi4-fix

closed 2020-03-31 by markondej