markondej/fm_transmitter #102: RPi4 not working

waterwin, 2020-01-17:
Tested with DMA channels 0 (default) as well as -d 7 and -d 14. Not OK.

There is some RF as the squelch of the receiver reacts, but no sound.

The same works on RPi3 with all 3 DMA channels. Is OK.

pcouderc12, 2020-03-13:
What is the situation about RPi4 ? Can I help ?
I have tried with no DMA too but no success.

markondej, 2020-03-21:
Please use branch https://github.com/markondej/fm_transmitter/tree/rpi4-fix (see Readme) and let me know if it works

pcouderc12, 2020-03-22:
Clearly it works !
at 90Mz without no more parameter and procedure that indicated !
But quality is fairly bad...

pcouderc12, 2020-03-22:
Trial on 88Mhz, with gregorian song from a commercial CD downloaded in mp3, (converted to wav with sox "-r 22050 -c 1 -b 16 -t wav ") shows fairly good quality but 2 strong noises, a residual noise due to the fact that the volume needs to be fairly high to be able to hear the song,  and a sharp sound, I would say a sine wave, but not so sharp as I hear it and I am old... When near of the rpi4, there is too some transmission  on about 92Mh, maybe stronger but noisy (saturated in fact).
pierre at couderc.eu. It disappears after a few meters.

markondej, 2020-03-23:
I don't which one exacly, but there is some hardware interfering with clock outputs on RPi4 (especially on GPIO4, which cannot be used to generate radio waves with higher frequency than 80 MHz anymore). Not sure if it is some kind of lowpass filter or if it comes from external device like Bluetooth antenna etc.
I'm still trying to figure out how to improve sound quality. Using spectrum analyzer I've figured out the distorsion is smaller on GPIO21, and also no signal is generated above 93.7 MHz.

pcouderc12, 2020-03-23:
OK, I am ready for more tests...
Do you think that fm_transmitter could be adapted for oranpi zero ? http://www.orangepi.org/orangepizero/

pcouderc12, 2020-04-18:
Thank you very much... I turn back to a raspberry pi zero w...

hex55, 2020-04-19:
> Please use branch https://github.com/markondej/fm_transmitter/tree/rpi4-fix (see Readme) and let me know if it works

It can't work, I test it. But GPIO4 can work for a while，about 10  secs, then corrupt.

goutx2, 2020-04-19:
hello they published  BCM2711 peripherals
https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bcm2711
i think it will help a lot especially in dma address i see there are 4 dma lite my be thise ise the probleme 
i dont know how to change the code of fm transmiter.ccp ''DMA channel ,addresse '' but all these information are described in ''BCM2711 peripherals.pdf ''
so please take a look at these docs 
We are waiting

goutx2, 2020-04-19:
> 
> 
> Tested with DMA channels 0 (default) as well as -d 7 and -d 14. Not OK.
> 
> There is some RF as the squelch of the receiver reacts, but no sound.
> 
> The same works on RPi3 with all 3 DMA channels. Is OK.

how to change dma channels ???

markondej, 2020-04-19:
> how to change dma channels ???

Use -d option to specify DMA channel (255 disables DMA, CPU will be used instead)

markondej, 2020-04-19:
> It can't work, I test it. But GPIO4 can work for a while，about 10 secs, then corrupt.

Please disable ARM CPU scalling governor and use GPIO21 on RPi4 (see README)

aaftabnaim, 2020-04-20:
Hi,I've tried all the steps mentioned in the description in order to get this running on my Pi 4B having no luck at all.The radio frequency gets some sort of input from the RPi but i don hear the sample song included being played.What else can i try to get this running?

goutx2, 2020-04-20:
Yesterday  i was playing in pi-fmrds.c code exactly in adresse 's (dma gpio, clk) using bcm2711 periferals.pdf doc as reference 
The signal was very  strong  i could listning 
Even without using any wire in gpio's and the sound quality was perfect 
I keep playing to get transmising above 93mhz but my rpi4 crash 
At these stage i dont know wich modification give me the strong signal and the perfect sound because i changed a lot of things in the code 
Now i cant get the the same prrfect transmition i have to start again 
As conclusion we can get strong signal with good quality "sorry for my bad English "

aaftabnaim, 2020-04-20:
@goutx2 Thanks a lot.First when i installed PiFmRds i got the file playing on my radio but there were some pulses in the background.
Then I tried -->  **echo "performance"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor**
That fixed the problem with the pulses.Now its working fine with a good range of about 20 meters.

goutx2, 2020-04-20:
> > It can't work, I test it. But GPIO4 can work for a while，about 10 secs, then corrupt.
> 
> Please disable ARM CPU scalling governor and use GPIO21 on RPi4 (see README)

> > It can't work, I test it. But GPIO4 can work for a while，about 10 secs, then corrupt.
> 
> Please disable ARM CPU scalling governor and use GPIO21 on RPi4 (see README)

In your code transmittter.cpp the adresse of gpio is GPIO_BASE_OFFSET 0x00200000
But in " bcm2711 peripherals.pdf"
The GPIO register base address is 0x7E21 5000
Can you explain please
I am trying  to make it work above 93mhz

markondej, 2020-04-21:
Peripherials are controlled by Peripherials singleton object initialized in this line:
https://github.com/markondej/fm_transmitter/blob/master/transmitter.cpp#L143

This object maps virtual memory to peripherials adress space starting from 0x7e000000, so if you want to access 0x7e2000000 you will need to access virual memory base + 0x00200000 (offset).

In pervious chip generations GPIO base address was 7e200000, I didn't know this has been changed.

LockeKK, 2020-05-25:
it is not recommenced to use performance governor in CPUFreq. it could workaround the issue but it will fix all the CPU cores at highest frequency.

goutx2, 2020-05-30:
> Peripherials memory is controlled by Peripherials singleton object initialized in this line:
> https://github.com/markondej/fm_transmitter/blob/master/transmitter.cpp#L143
> 
> This object maps virtual memory region to peripherials adress space starting from 0x7e000000, so if you want to access 0x7e2000000 you will need to access virual memory base + 0x00200000 (offset).
> 
> In pervious chip generations GPIO base address was 7e200000, I didn't know this has been changed.

It is  correct gpio adresse  is 07e200000
 there's a mistake in that document
https://github.com/raspberrypi/documentation/issues/1569