markondej/fm_transmitter #113: Play from a bluetooth source

JohnMatychuk, 2020-04-24:
Is there any way to play audio files from my phone to the pi via bluetooth and then transmit to FM?

Rustizx, 2020-05-23:
I'm trying to do this samething. I'm attempting to use this https://github.com/lukasjapan/bt-speaker and configure it in a way that will send the output of the bluetooth audio to the fm_trasmitter. I am using a ALSA loopback in order to connect the two. Hopefully this works for you, I'm still trying to get it to work myself.

lguima, 2020-06-12:
I found this channel in YouTube: [Novaspirit Tech](https://www.youtube.com/channel/UCrjKdwxaQMSV_NDywgKXVmw)

there's 2 of 3 videos related to "Bluetooth Audio FM Transmitter for Raspberry Pi" project, but it seems he never finished :(

[Bluetooth Audio FM Transmitter for Raspberry Pi 3/2 Part 1](https://www.youtube.com/watch?v=vR0ullht9VU)
[Bluetooth Audio FM Transmitter for Raspberry Pi 3/2 Part 2](https://www.youtube.com/watch?v=tGQKjeeNNSw)

andreaslordos, 2021-01-22:
> I'm trying to do this samething. I'm attempting to use this https://github.com/lukasjapan/bt-speaker and configure it in a way that will send the output of the bluetooth audio to the fm_trasmitter. [...]

@Rustizx Did you find a way to make it work?