markondej/fm_transmitter #114: can't compile on raspberry pi 4 (aarch64, gcc 9.2)

xlla, 2020-05-13:
<pre>
rpi4:~/git/fm_transmitter$ uname -a
Linux rpi4 4.19.115 #1 SMP PREEMPT Tue Apr 21 10:36:16 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
rpi4:~/git/fm_transmitter$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/aarch64-poky-linux/9.2.0/lto-wrapper
Target: aarch64-poky-linux
Configured with: ../../../../../../work-shared/gcc-9.2.0-r0/gcc-9.2.0/configure --build=x86_64-linux --host=aarch64-poky-linux --target=aarch64-poky-linux --prefix=/usr --exec_prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/libexec --datadir=/usr/share --sysconfdir=/etc --sharedstatedir=/com --localstatedir=/var --libdir=/usr/lib --includedir=/usr/include --oldincludedir=/usr/include --infodir=/usr/share/info --mandir=/usr/share/man --disable-silent-rules --disable-dependency-tracking --with-libtool-sysroot=/ --with-gnu-ld --enable-shared --enable-languages=c,c++,fortran --enable-threads=posix --enable-multilib --enable-default-pie --enable-c99 --enable-long-long --enable-symvers=gnu --enable-libstdcxx-pch --program-prefix=aarch64-poky-linux- --without-local-prefix --enable-lto --disable-libssp --enable-libitm --disable-bootstrap --disable-libmudflap --with-system-zlib --with-linker-hash-style=gnu --enable-linker-build-id --with-ppl=no --with-cloog=no --enable-checking=release --enable-cheaders=c_global --without-isl --with-build-sysroot=/ --disable-static --enable-nls --with-glibc-version=2.28 --enable-initfini-array --enable-__cxa_atexit
Thread model: posix
gcc version 9.2.0 (GCC) 

</pre>
make will stop for loses precision error:
<pre>
rpi4:~/git/fm_transmitter$ make GPIO21=1
g++ -Wall -O3 -std=c++11 -fno-strict-aliasing -I/opt/vc/include -DGPIO21 -c transmitter.cpp
transmitter.cpp: In member function ‘uint32_t Peripherals::GetPhysicalAddress(volatile void*) const’:
transmitter.cpp:131:78: error: cast from ‘volatile void*’ to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  131 |             return PERIPHERALS_PHYS_BASE + (reinterpret_cast<uint32_t>(object) - reinterpret_cast<uint32_t>(peripherals));
      |                                                                              ^
transmitter.cpp:131:120: error: cast from ‘void*’ to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  131 |             return PERIPHERALS_PHYS_BASE + (reinterpret_cast<uint32_t>(object) - reinterpret_cast<uint32_t>(peripherals));
      |                                                                                                                        ^
transmitter.cpp: In member function ‘uint32_t Peripherals::GetVirtualAddress(uint32_t) const’:
transmitter.cpp:134:58: error: cast from ‘void*’ to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  134 |             return reinterpret_cast<uint32_t>(peripherals) + offset;
      |                                                          ^
transmitter.cpp: In member function ‘uint32_t AllocatedMemory::GetPhysicalAddress(volatile void*) const’:
transmitter.cpp:195:79: error: cast from ‘volatile void*’ to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  195 |             return (memSize) ? memAddress + (reinterpret_cast<uint32_t>(object) - reinterpret_cast<uint32_t>(memAllocated)) : 0x00000000;
      |                                                                               ^
transmitter.cpp:195:122: error: cast from ‘void*’ to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  195 |             return (memSize) ? memAddress + (reinterpret_cast<uint32_t>(object) - reinterpret_cast<uint32_t>(memAllocated)) : 0x00000000;
      |                                                                                                                          ^
transmitter.cpp: In member function ‘uint32_t AllocatedMemory::GetAddress() const’:
transmitter.cpp:198:59: error: cast from ‘void*’ to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  198 |             return reinterpret_cast<uint32_t>(memAllocated);
      |                                                           ^
transmitter.cpp: In member function ‘void Transmitter::TransmitViaDma(WaveReader&, ClockOutput&, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int)’:
transmitter.cpp:438:94: error: cast from ‘volatile DMAControllBlock*’ to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  438 |     volatile uint32_t *clkDiv = reinterpret_cast<uint32_t *>(reinterpret_cast<uint32_t>(dmaCb) + 2 * sizeof(DMAControllBlock) * bufferSize);
      |                                                                                              ^
transmitter.cpp:439:100: error: cast from ‘volatile uint32_t*’ {aka ‘volatile unsigned int*’} to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  439 |     volatile uint32_t *pwmFifoData = reinterpret_cast<uint32_t *>(reinterpret_cast<uint32_t>(clkDiv) + sizeof(uint32_t) * bufferSize);
      |                                                                                                    ^
make: *** [makefile:22: transmitter.o] Error 1
</pre>

If I add -fpermissive to Makefile, compile fine, but failure on run.
<pre>
rpi4:~/git/fm_transmitter$ sudo ./fm_transmitter -f 92 acoustic_guitar_duet.wav 
Broadcasting at 92 MHz with 200 kHz bandwidth
Playing: acoustic_guitar_duet.wav, 22050 Hz, 16 bits, mono
Segmentation fault

</pre> 

should I change those "reinterpret_cast<uint32_t>" to "uintptr_t" ?

xlla, 2020-05-13:
I have changed some uint32_t to uintptr_t by study pull#91, it now works.

closed 2020-05-13 by xlla

QuentinDel, 2020-12-14:
Hello, 
Could you please make your code available? 

Best regards,
Q