markondej/fm_transmitter #116: Informations about output signal in antenna

Nashento, 2020-07-05:
Hello Markondej ! 

First of all : thank you so much for your code ! It's great and it works fine for me. I tried it with a bad and small antenna of 20 cm and i got à range of 30m (average in city) with a raspberry pibow model B and frequency = 100MHz.

I have 2 little questions cause I'm an electronic engineer and i work on a FM amplifier : did you see the output signal on GPIO4 ? (Is there a signal between +5V and -5V ? Or another things ?)
The next is : is there a limit about minimum frequency ? Like : 80MHz ? 20 MHz ? ...

Thank you for your response and thank you again for your job !

Nashento 

My email if you want to reply by this way : hironkx@gmail.com

markondej, 2020-07-05:
Hi, I haven't egzamined gpio signal directly due to lack of such precise osciloscope (mine has resolution only up to 20 MHz). I beleive it should look like regular rectangular wave with amplitude of 3.3V with minimum value of 0V, not negative (as this GPIO supports only 3.3V input and output signals).

Supported band needs to be checked empirically. With simple SDR spectrum analyzer I've found it working nicely with 60MHz signal also. As far as I remember I've also tried to transmit on PMR band (416MHz) but this was to high.

Nashento, 2020-07-07:
Hi ! 
Ok, thank you for your response. I'm going to make some test to analyse it. Thank you again.