markondej/fm_transmitter #121: Error while running "Make" command

Sabering1, 2020-09-14:
Hi i was just downloading this and while i was running a "make" command i got a error like this:
[image]
[image]

markondej, 2020-09-15:
Looks like broadcom library is missing or badly linked. Please specify your OS (Kali Linux, am I right?) and gcc version.

Sabering1, 2020-09-15:
I am using Kali Linux 2020.3 on my RP zero w and the gcc version is (Debian 9.2.1-25) 9.2.1 20200123
Here is the screenshot:
[image]

Bl00dyDevil, 2021-01-04:
I have the same problem. I'm running Kali linux 2020.4 on RPi4. here is the log: 

```
g++ -Wall -O3 -std=c++11 -DVERSION=\"0.9.4\" -DEXECUTABLE=\"fm_transmitter\" -c main.cpp
g++ -Wall -O3 -std=c++11 -c mailbox.c
g++ -Wall -O3 -std=c++11 -c sample.cpp
g++ -Wall -O3 -std=c++11 -c wave_reader.cpp
g++ -Wall -O3 -std=c++11 -fno-strict-aliasing -I/opt/vc/include -c transmitter.cpp
transmitter.cpp:36:10: fatal error: bcm_host.h: No such file or directory
   36 | #include <bcm_host.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [makefile:22: transmitter.o] Error 1
```
Btw yes i have root permissions.

markondej, 2021-03-16:
I've updated the readme, please try installing Broadcom packages first: libraspberrypi-dev  libraspberrypi-bin

closed 2021-03-16 by markondej