markondej/fm_transmitter #122: [NOT AN ISSUE] Playing not just one music

Aqua-22, 2020-09-21:
Hello i am wondering if i can have a folder in the fm_transmitter folder whit a lot of music and have them play after the other.

closed 2020-12-16 by Aqua-22

superamario64, 2021-02-20:
Did you ever figure out how to do this?

Markieyer, 2021-03-01:
You can use a wildcard to do all music, so instead of putting "/home/Music/My\ Song.wav", you'd put "/home/Music/*.wav"
It'll do it all in alphabetical order, keep that in mind, I've yet to find a way to do a truly random or, 'shuffle' with it yet, but still, best way of playing all your music is like that

superamario64, 2021-03-01:
oh great! thanks a lot
one way you could shuffle the songs is to maybe write a simple script that renames all the files to `001.wav`, `002.wav` in a random way