markondej/fm_transmitter #123: i can't install fm_trasmitter

BOT-CODER, 2020-10-30:
```
g++ -L/opt/vc/lib -lm -lpthread -lbcm_host -o fm_transmitter main.o mailbox.o sample.o wave_reader.o transmitter.o
/usr/bin/ld: transmitter.o: in function `Transmitter::TransmitterThread(Transmitter*, ClockOutput*, unsigned int, unsigned int, unsigned int, unsigned int*, std::vector<Sample, std::allocator<Sample> >*)':
transmitter.cpp:(.text+0x404): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x414): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x424): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.o: in function `Transmitter::TransmitViaCpu(WaveReader&, ClockOutput&, unsigned int, unsigned int, unsigned int, unsigned int)':
transmitter.cpp:(.text+0x8ec): undefined reference to `pthread_create'
/usr/bin/ld: transmitter.o: in function `Transmitter::TransmitViaDma(WaveReader&, ClockOutput&, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int)':
transmitter.cpp:(.text+0x970): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x97c): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.cpp:(.text+0xa24): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0xa30): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.cpp:(.text+0x1100): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1110): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1120): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.cpp:(.text+0x1258): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1268): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1278): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.cpp:(.text+0x1318): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1328): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1338): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.o: in function `Transmitter::Transmit(WaveReader&, float, float, unsigned int, bool)':
transmitter.cpp:(.text+0x17fc): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1808): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.cpp:(.text+0x182c): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1838): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.cpp:(.text+0x1ab4): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1ac4): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.cpp:(.text+0x1ad0): undefined reference to `bcm_host_get_peripheral_address'
/usr/bin/ld: transmitter.o: in function `Peripherals::~Peripherals()':
transmitter.cpp:(.text._ZN11PeripheralsD2Ev[_ZN11PeripheralsD5Ev]+0x24): undefined reference to `bcm_host_get_peripheral_size'
/usr/bin/ld: transmitter.o: in function `Peripherals::~Peripherals()':
transmitter.cpp:(.text._ZN11PeripheralsD0Ev[_ZN11PeripheralsD5Ev]+0x24): undefined reference to `bcm_host_get_peripheral_size'
collect2: error: ld returned 1 exit status
make: *** [makefile:10: all] Error 1
```

i am using raspberry pi zero with pw4n p1 aloa kali linux OS. please tell me the way to fix it.

BOT-CODER, 2020-10-30:
waiting for your reply

Greatest125, 2020-12-12:
Perhaps try with Raspbian installed? Is the error above when you run `sudo make`?

rsaxvc, 2021-02-27:
Have you seen #104 ? You will need the broadcom libraries installed and available in the linker path.

Here's how I check if those functions exist in the broadcom library on RPiOS:

`
pi@raspberrypi:~/code/fm_transmitter $ nm /opt/vc/lib/libbcm_host.so  | grep bcm_host_get
00003e14 T bcm_host_get_model_type
00003b1c T bcm_host_get_peripheral_address
00003c38 T bcm_host_get_peripheral_size
00003f0c T bcm_host_get_processor_id
00003d70 T bcm_host_get_sdram_address
pi@raspberrypi:~/code/fm_transmitter $
`

If this returns similar entries for the bcm_host_ functions, you might try reordering the linker flags like was done in #104