markondej/fm_transmitter #126: Building on Ubuntu Server 20.10

Andreas711, 2020-12-16:
Hello,

I recently installed Ubuntu Server 20.10 64bit on my raspberry pi 3 b rev 1.2.

When I go to "sudo make", I get this error

<<
g++ -Wall -O3 -std=c++11 -DVERSION=\"0.9.4\" -DEXECUTABLE=\"fm_transmitter\" -c main.cpp
g++ -Wall -O3 -std=c++11 -c mailbox.c
mailbox.c: In function ‘void* mapmem(unsigned int, unsigned int)’:
mailbox.c:64:33: error: cast from ‘void*’ to ‘int’ loses precision [-fpermissive]
   64 |       printf("mmap error %d\n", (int)mem);
      |                                 ^~~~~~~~
make: *** [makefile:13: mailbox.o] Error 1
>>

Someone please help!

Andreas

Andreas711, 2020-12-16:
I had a look at a previous Issue and found that I should add "FLAGS = -Wall -O3 -std=c++11 -fpermissive" to line 3 of the makefile. I did this, and it makes more progress but fails again...

*Here is the error*
<<
g++ -Wall -O3 -std=c++11 -fpermissive -c mailbox.c
mailbox.c: In function ‘void* mapmem(unsigned int, unsigned int)’:
mailbox.c:64:33: warning: cast from ‘void*’ to ‘int’ loses precision [-fpermissive]
   64 |       printf("mmap error %d\n", (int)mem);
      |                                 ^~~~~~~~
g++ -Wall -O3 -std=c++11 -fpermissive -c sample.cpp
g++ -Wall -O3 -std=c++11 -fpermissive -c wave_reader.cpp
g++ -Wall -O3 -std=c++11 -fpermissive -fno-strict-aliasing -I/opt/vc/include -c transmitter.cpp
transmitter.cpp:36:10: fatal error: bcm_host.h: No such file or directory
   36 | #include <bcm_host.h>
        |                ^~~~~~~~~~~~
compilation terminated.
make: *** [makefile:22: transmitter.o] Error 1
>>

themoonwatcher, 2020-12-16:
It looks like you have to find the sources of the Rpi's userland files and fill the /opt/vc/include directory.

Andreas711, 2020-12-17:
How do I go about doing this?

dengjy-vi, 2021-01-25:
I have the same problem

qxliang2014, 2021-02-23:
I have the exactly same issue on my RaspberryPi 4b with Ubuntu Serve 20.10,still trying to fix it...

rsaxvc, 2021-02-27:
I think there's a better way than -fpermissive. I'll put up a patch.

But you also need the Broadcom headers and libraries. For RPiOS you would do this if they are not installed,
`sudo apt-get install libraspberrypi-dev raspberrypi-kernel-headers` , but I'm not sure about Ubuntu Server 20.10.

markondej, 2021-03-16:
Works fine after @rsaxvc fixes

closed 2021-03-16 by markondej