markondej/fm_transmitter #127: Music running very slow when broadcasting

Aqua-22, 2020-12-16:
I am running this on my raspberry pi 3 and when i broadcast to any frequency the music is playing like half the speed so very slow.
What can be the issue?

CapCarCap, 2020-12-17:
I am having the same issue when live broadcasting with arecord on pi 3a+

markondej, 2020-12-20:
Please check and provide me used sampling rate, there may be issues with PWM controller clock frequency.

kzl86, 2021-01-15:
This is the issue on my Raspberry Pi 3A+ playing the example acoustic_guitar_duet.wav. It's around half the speed compared to playing the same wav file with omxplayer. Using Raspbian GNU/Linux 10 (buster). The source code was complied with make without any modification.

danilalugovoy, 2021-01-17:
I have the same problem, Raspberry Pi 3. The guitar example works, but when I try to put a custom wave file everything breaks. I used to do the same and everything worked. The sampling rate is 22050.

alinmindroc, 2021-01-30:
I had the same problem on a Pi 3, and found out that if you connect an HDMI display, for some reason the sound is played at the right speed. 
Probably caused by some clock speed change when there is high load on the Pi, like this issue describes: https://github.com/markondej/fm_transmitter/issues/101 

Anyway, in my case it looks like if you disable DMA it will play the sound at the right speed always, so worth a try for others:
`sudo ./fm_transmitter -f 100 -d 255 acoustic_guitar_duet.wav`

kzl86, 2021-01-30:
Simply connecting a HDMI display hasn't helped, neither setting the CPU on a higher frequency with:
echo performance > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor

However, the -d 255 did the trick:
sudo ./fm_transmitter -f 87.5 -d 255 acoustic_guitar_duet.wav

Now I'm able to play music, frequency stays 600Mhz (governor: ondemand) without external display.
@alinmindroc Thank you very much for your help!

Cheers

closed 2021-06-21 by Aqua-22

derf-673, 2022-03-02:
How do I disable the DMA?