markondej/fm_transmitter #129: Raspberry 4 not working

Kisiekx, 2021-01-19:
Hello,
I made everything like you said. I installed it and then I put this commands:
make GPIO21=1
echo "performance"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
then I played example audio: sudo ./fm_transmitter -f 92.0 acoustic_guitar_duet.wav
and it says that it is broadcasting but it's not.
I'm using RPi4b 8gb ram and i read this: https://github.com/markondej/fm_transmitter/issues/102 but nothing did work.
Do you have any idea why it is not working?

CatalinPuscoci, 2021-01-21:
Try lower frequencies, and I strongly suggest attaching a wire to act as an antenna to that gpio pin if you want to hear it properly. I found myself sticking to 50MHz more often than not.

More details about your setup would be nice too.

Kisiekx, 2021-01-21:
I'm using raspberry pi 8gb ram and raspbian. Actually i found out that although i used: make GPIO21=1 not GPIO21 was transmitting but GPIO4. I plugged wire to GPIO4 and it was working but very bad. I tried plugging wire to GPIO21 but nothing was happening.

TheLiuzhu, 2021-05-24:
When I first install the code, it don't works  at all either. However, after one day(tmr morning), it works
PS: This is real ,I am NOT kidding, so maybe wait will help (at least on my RPi4B 4G).