markondej/fm_transmitter #132: It won't make the file

stubbertville, 2021-02-11:


rsaxvc, 2021-02-27:
What version of g++ do you have installed?

stubbertville, 2021-02-27:
> What version of g++ do you have installed?

I'm new to pi os, or Linux in that case, whats that?

rsaxvc, 2021-02-27:
TLDR: Your compiler is probably out of date and likely needs upgraded.

Background: g++ is a compiler for the C++ language. c++11 is a specific version of the C++ language, but it's been around for quite a while. "-std=c++11" is a flag to tell g++ which version of the c++ language standard to use. But, "unrecognized command Line option '-std=c++11'" here means g++ did not understand the specified language version, which usually means g++ needs to be updated.

You can find the g++ version by adding the --version flag, just like this "g++ --version". Here's the result from one of my RPi4 for reference:

`
pi@raspberrypi:~/code/fm_transmitter $ g++ --version
g++ (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pi@raspberrypi:~/code/fm_transmitter $
`

Since the compiler is usually packaged by the same group providing the OS, you might also check your OS version, here's mine:

`
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $
`

After reviewing the [Raspberry Pi OS Release History](https://en.wikipedia.org/wiki/Raspberry_Pi_OS#Release_history), it's quite confusing to me, because even the first Raspberry Pi OS came with GCC/G++ 4.7.2, [which didn't support everything but at least knew about the C++11 language version](https://gcc.gnu.org/gcc-4.7/cxx0x_status.html).

AlienX51, 2022-02-25:
I'm trying to install fm_trasmitter software for my raspberry pi 3, I'm using kali as the OS (since it has a little bit of everything already in the OS and don't need to install majority of the stuff ), I'm trying to compile the code using the line sudo make. and this is what I get:

NOTE: I have updated everything

g++ -Wall -O3 -std=c++11 -DVERSION=\"0.9.6\" -DEXECUTABLE=\"fm_transmitter\"

g++ -Wall -O3 -std=c++11 -c mailbox.c

g++ -Wall -O3 -std=c++11 -c wave_reader.cpp

g++ -Wall -O3 -std=c++11 -fno-strict-aliasing -I/opt/vc/include -c transmitte

transmitter.cpp:36:10: fatal error: bcm_host.h: No such file or directory

36 | #include <bcm_host.h>

| ^~~~~~~~~~~~

compilation terminated.

make: *** [makefile:19: transmitter.o] Error 1

what can it be..????