markondej/fm_transmitter #133: Raspberry Pi 4: FM Transmitter doesn't work on reboot

RickBNK, 2021-02-25:
Hi all,

first of all: it's a great project and really neatly done! I appreciate your work on this!

The transmitter is also working very well on my rpi4, however only on the first start.
The first time I installed the fm_transmitter it worked like a charm, even on GPIO4. I could start and stop transmitting without issues with the standard commands.

I then rebooted my rpi4 to check that my auto start functionality works (transmitting immediately when I boot).
Unfortunately it didn't. I went into the console and tried starting it from there, but this also didn't work. It said it was transmitting but nothing was picked up by the same radio on the same frequency as before.
Additional reboots didn't work either until I gave up, shut down and unplugged the rpi4 from the power source.

The next day I tried again and again it worked on the initial start up - like a charm! Nothing after a reboot though. It always says it's transmitting but nothing is picked up.

I now somewhat figured out that, depending on how long I was transmitting, I need to shut down the rpi4 and have the power unplugged for either 10 minutes or up to 2 hours. Starting it up again afterwards and it works.

I can't seem to figure out why this is the case. I switched it to GPIO21 and had the same effect (working on "clean" start, nothing on reboot or similar).

I also tried the performance setting but no success either.

`echo "performance"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor`

Do you have any idea what the root cause could be and how I can fix it?

Thanks!

Rick

rsaxvc, 2021-02-27:
I think I've also seen this. I plan to connect it to my oscilloscope and pull out my RTLSDR to be sure because I haven't been able to get stable audio out yet, but sometimes I cannot even generate static until the Pi is fully powered off. Perhaps it has something to do with the DMA controller state?

markondej, 2021-03-02:
RPi 4 hardware is problematical, I've noticed it's behaivour varies between firmware versions very often, esspecially when it comes to clocks etc. I will examine this issue with latest firmware ale will let You know if I come up with something.

RickBNK, 2021-03-03:
That's great, @markondej.

I ran 

`sudo rpi-eeprom-update`

on my rpi4 and that's the result, in case it helps:

> BCM2711 detected
> VL805 firmware in bootloader EEPROM
> Checking for updates in /lib/firmware/raspberrypi/bootloader/default
> Use raspi-config to select either the default-production release or latest update.
> BOOTLOADER: up-to-date
> CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
>  LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
> RELEASE: default
> VL805: up-to-date
> CURRENT: 000138a1
>  LATEST: 000138a1

rsaxvc, 2021-03-06:
I've added some tools here to monitor the peripheral registers when they change during operation. I only have a Pi4 set up currently, but I have a Pi B to try too.