markondej/fm_transmitter #143: music is really really quiet on RPi4

rullinoiz, 2021-07-18:
I've been having issues with this script all week, and I just got it to play something but I had to turn up the volume a whole ton. I compiled it with GPIO21=1 and I'm broadcasting to an unused radio frequency (87.9). Boosting the audio of the wav does not work because when I play it on a music player it's just as loud as it should be. Is there any way to get at least any volume boost on the pi 4?

rullinoiz, 2021-07-18:
i found out turning up the frequency i was broadcasting at made the quality better, as well as the -b parameter up to 100000 makes it louder. so far the quality isnt as great as playing on a RPi3 model b, but i can still make out the music

Fordi, 2021-07-29:
A `-b` greater than 200 is a violation of the spec (FM stations have a bandwidth of 180-200 kHz - 180 should really be the default).  I've had best results on a Pi400 running on 87.9 MHz / ±180kHz with a 44.1 kHz stereo source, but it's still crackly at a 2ft distance.