markondej/fm_transmitter #144: [NOT AN ISSUE, QUESTION] How to transmit the audio of the raspberry (not a file, not a microphone)

ImHumbug, 2021-07-28:
Hi so I'm using a music bot that plays songs from youtube into the raspberry, so if I plug headphones to it I can listen songs. Is there any way I can transmit the audio played in the raspberry to fm? I could't find any parameters to do that.

hydranix, 2021-08-04:
Here's what I do.

First I load the alsa loopback modile.

`modprobe snd-aloop`

This creates a loopback audio device.

Then I use the arecord program.  (part of alsa-utils package)

I pipe the output of arecord to fm_transmitter like this:

`arecord -D hw:0,1,0 -c 1 -d 0 -r 44100 -f S16_LE|fm_transmitter -f 102.5 -`

Now just play your audio through the loopback device. Be sure to use subchannel 0

`aplay -D hw:0,0,0 -c 1 -r 44100 -f S16_LE /path/to/audio.file`

Or you use arecord your microphone and pipe it into aplay to loopback device. You can also do some more interesting things like mix your mic into the playing audio file. Kind of like a live DJ...

markondej, 2022-02-08:
I've added these informations in readme file

closed 2022-02-08 by markondej

fuadxtps23, 2023-06-20:
> Hi so I'm using a music bot that plays songs from youtube into the raspberry, so if I plug headphones to it I can listen songs. Is there any way I can transmit the audio played in the raspberry to fm? I could't find any parameters to do that.

can i have the youtube music bot?

fuadxtps23, 2023-06-22:
> Here's what I do.
> 
> First I load the alsa loopback module.
> 
> `modprobe snd-aloop`
> 
> This creates a loopback audio device.
> 
> Then I use the arecord program. (part of alsa-utils package if you use ArchLinux)
> 
> I pipe the output of arecord to fm_transmitter like this:
> 
> `arecord -D hw:0,1,0 -c 1 -d 0 -r 44100 -f S16_LE|fm_transmitter -f 102.5 -`
> 
> Now just play your audio through the loopback device. Be sure to use subchannel 0
> 
> `aplay -D hw:0,0,0 -c 1 -r 44100 -f S16_LE /path/to/audio.file`
> 
> Or you use arecord your microphone and pipe it into aplay to loopback device. You can also do some more interesting things like mix your mic into the playing audio file. Kind of like a live DJ...

not working for me, but working for me is this command
`arecord -c 2 -f S16_LE -r 44100 | sudo ./fm_transmitter -f 98.6 -d 255 -`

alheyne, 2023-11-15:
So if I wanted to play a movie file(for projection mapping to my house) how might I re-direct the audio out from vlc media player to fm_transmitter?  Any help would be greatly appreciated.  thanks in advance.

hydranix, 2023-11-23:
> ayer to fm_transmitter? Any help would be greatly appreciated. thanks in advance.

It's pretty simple. Scroll up to my comment about using an alsa loopback device.

Then just point VLC's audio output to the loopback device and thats it. 

`vlc --alsa-audio-device=hw:0,0,0 --alsa-audio-channels=2  <rest of your vlc commandline>`

I haven't tested the above command, but that should give you enough info to get it going.