markondej/fm_transmitter #146: How to add music

luke1310, 2021-08-30:
Hey guys, I'm just learning a lot about FM Radio and the Raspberry. Could someone please tell me how to add the wav. Files to the pi? I know it is explained in the read me, but i don't get it :(

markondej, 2021-09-01:
WAV is simply sound file format, used since... well I'm sure it was used by Windows 95. To get such file on your Pi you can eg. download it from internet by using wget command in terminal:
```
wget http://some-music-source.com/download/example.wav
```
Another way is to upload WAV file to the Pi with SCP client program (built in linux or by using WinSCP on Windows machine) simply passing your Pi's SSH credentials (SSH must ve enabled - see foundqtion tutorials how to).

Please notice only raw WAV files are supported by transmitter itself, eg. some of them make use of compression and this won't work. You can always use ANY music format (including compressed WAV files) by converting it to either STDOUT stream or raw WAV (see readme).

luke1310, 2021-09-05:
So like this?
scp <Path to File To Copy> pi@<IP Address of Raspberry Pi>:<Path that File will Go>

But where should the music files go?
Should i create a new folder e.g.:
Mkdir music
Cd music
 And then:
scp <Path to File To Copy> pi@<IP Address of Raspberry Pi>: home/pi/music

?

luke1310, 2021-09-05:
For some reason, it deleted the scp command. 
Is it : scp Path to File To Copy pi@IP Address of Raspberry Pi:Path that File will Go ?

luke1310, 2021-09-05:
Thanks it works for me👍.

closed 2021-09-05 by luke1310