markondej/fm_transmitter #147: I can't receive it from the radio.

jay9455, 2021-09-15:
Even though I followed all the guidelines on github, it is not playing normally.

I connected the cable like the picture below to GPIO4, GPIO21, but when I set the frequency on the radio, only noise is heard.

Is there any way to solve it?

[image]

gitiliya, 2021-09-23:
Im still tinkering around with mine but if you use -b to make it louder you can actually begin to hear it, sudo ./fm_transmitter -f 87.60 -b 3000 rager.wav

KodeNoleg, 2021-09-28:
What are you putting in the terminal to play the music?

jay9455, 2021-09-30:
> 음악을 재생하기 위해 터미널에 무엇을 넣습니까?

`sudo ./fm_transmitter -f 102.0 acoustic_guitar_duet.wav`

I used the command above.

However, when Ubuntu was reinstalled, it worked normally.

closed 2021-09-30 by jay9455