markondej/fm_transmitter #148: Transmitting Is Too Weak/Quiet

gitiliya, 2021-09-23:
Running the transmitter normally makes no sound, when I turn up the volume to 5000 is when I begin to actually hear the audio come into my FM radio receiver. Also using GPIO pin 21 doesn't seem to work at all.

Chanakan55991, 2021-10-03:
the reason that it's leaking to other frequencies is because you need a band pass filter.

ranfur8, 2021-11-25:
I'm having the same issue, not the leakage, but the low power, I have a 30 cm jumper cable on GPIO pin 7, as some people say that should give me about 10m range, I barely get a few centimetres. Am I doing something wrong? Is my raspberry pi model not compatible? (Pi4 B 8GB) Is it because the frequency spectrum is too crowded and the noise is interfering with the transition to the point where it barely transmits at a few centimetres?   [Image Reference](https://imgur.com/a/XCsS8Tk)

ranfur8, 2021-12-09:
> I'm having the same issue, not the leakage, but the low power, I have a 30 cm jumper cable on GPIO pin 7, as some people say that should give me about 10m range, I barely get a few centimetres. Am I doing something wrong? Is my raspberry pi model not compatible? (Pi4 B 8GB) Is it because the frequency spectrum is too crowded and the noise is interfering with the transition to the point where it barely transmits at a few centimetres? [Image Reference](https://imgur.com/a/XCsS8Tk) (I used the pin 40 as you said, did the make GPIO21=1 but it stopped working at all so i just rolled back to pin 7 setup)

I think I figured it out. 
If you are using a jumper cable, one of those can come with kits and such, those are utter garbage. I took a 30 cm solid core copper cable and soldered it to GPIO PIN 7 on the board and now it works perfectly, up to 10m range in open field, 5m range in closed areas.
If you dont want to solder it to the board, you can use a female header and attach the cable to it, but you might run into the same issue again.

CookieGMVN, 2022-04-04:
Pi 4B is very weak in FM transmitter, I don't know why (maybe 4B have different built-in hardware like [README.md](https://github.com/markondej/fm_transmitter/blob/master/README.md) said). My Pi 4B can transmit just some centimeters (about 30-50 cm). If you want better FM transmit, take a look at Pi Zero. The Pi Zero is very good and it can transmit around 3-4 m away the location where you placed your Pi.

BobHasNoSoul, 2022-04-16:
> per cable, one of those can come with kits and such, those are utter garbage. I took a 30 cm solid core copper cable and soldered it to GPIO PIN 7 on the board and now it works perfectly, up to 10m range

can confirm jumper cable is useless solder onto the pin did that and have near perfect signal all over my house and garden +1