markondej/fm_transmitter #152: live transmitting?

wickermonkey, 2021-12-19:
I know this probably wont get read as its an old project but its worth a try. ive got the mp3 transmitting using a zero pi 1.3 (none w) and was wondering if it was at all possible to get this working for general audio output of the zero pi? want to use it as a transmitter for youtube music/phone/anything else plugged in or playing (spotify/amazon.......). maybe use the io to be able to change the frequency on the fly.

dgaray01, 2022-01-28:
I would like to do this too! Did you got any solutions?

wickermonkey, 2022-01-29:
Unfortunately no. I think that there are add in boards that will be able to do what is needed but that wasn't what I was trying to achieve, so I haven't looked any further into it

markondej, 2022-02-03:
Please see: #144