markondej/fm_transmitter #153: Stereo Support needed.

p0lyX0r, 2021-12-19:
First of all I want to share my experience for turning the raspberry pi into FM transmitter aka PiFM, I tried almost all method to make PiFM but it ended up with no sounds transmitted. And only your project works on my Raspberry Pi 3 B.  I salute you man for such a brilliant way to come up with this project.
I have no complain of any issues because it works for me pretty smooth, loud and clear. I hope you will release the update to support stereo format.

Br,

markondej, 2022-02-03:
Will try to add subsampling and stereo support with some free time available.

Albonycal, 2022-07-05:
+1 stereo support would be nice

HackerDaGreat57, 2023-07-18:
Stereo has been implemented AFAIK. Tried it out yesterday and it's crisp as hell.