markondej/fm_transmitter #156: command not working with python os module

nathan3453, 2022-01-02:
i'v been creating this app with python os and tkinter but when I try to use the command through python it doen't work because ./fm_transmitter isn't a command

markondej, 2022-01-03:
Have you tried:
```
sudo install ./fm_transmitter /usr/local/sbin
```
after compiling this app?

nathan3453, 2022-01-03:
just used that don't think it helped same error message here's my code

from tkinter import *
import os

def station():
    
    def song1():
       stream = os.popen("sudo ./fm_transmitter -f 102.10 /home/pi/fm_transmitter/radio_station/songs/acoustic_guitar_duet.wav")
       output = stream.readlines()
       print(output)
  
  
   # def song2():
        
    #def song3():
        
    #def song4():
        
    #def song5():
        
        
    window = Tk()
    window.geometry("630x480")
    window.configure(background="white")
    label1 = Label(window, text="songs", bg="grey", padx=75, pady=10)
    button1 = Button(window, text="acoustic guitar duet", padx=75, pady=10, command=song1)
    button2 = Button(window, text="acoustic guitar duet", padx=75, pady=10)
    button3 = Button(window, text="acoustic guitar duet", padx=75, pady=10)
    button4 = Button(window, text="acoustic guitar duet", padx=75, pady=10)
    button5 = Button(window, text="acoustic guitar duet", padx=75, pady=10)
    
    
    

    
    
    # grid layout
    label1.grid(row=0, column=0)
    button1.grid(row=1, column=0)
    button2.grid(row=2, column=0)
    button3.grid(row=3, column=0)
    button4.grid(row=4, column=0)
    button5.grid(row=5, column=0)

    window.mainloop()