markondej/fm_transmitter #159: FRS two-way-radios?

ulisesvina, 2022-01-20:
I'm trying to transmit (NOT RECEIVE) audio with my Raspberry Pi using this program to a FRS two-way-radio (Midland LXT114) on it's channel 1 (462.562MHz freq, 12.5KHz bandwidth), when I approach the radio to the Raspberry Pi's antenna, it does pick up something, but it's just noice. I can't seem to play anything at all through it, even the example `acoustic_guitar_duet.wav` file has the same result.

markondej, 2022-02-03:
462 MHz with 12.5 KHz bandwidth is way too high frequency for RPi's clock hardware, it simply does not have sufficient resolution to transmit analog sound on such high frequency.

ulisesvina, 2022-02-08:
Should have closed this issue before, but thanks. Anyways, I managed to make it work (yes, even being such high frequency). My problem was that those kind of radios does not use FM, but NBFM (narrow-band FM). But thanks a lot anyway.

closed 2022-02-08 by ulisesvina